rstrouse/relayEquipmentManager #75: Sequent 4rel4in v4 - Feeds for digital input not working

brentk7, 2024-06-28:
I have a Sequent 4rel4in v4 card and have tried to configure the inputs to toggle a service mode.
As soon as I enable the feed, I start getting errors in the log as shown below. I have tried using both inDigital1.value and  plain inDigital1 as I am not sure what the difference is between them - but they both give the same error.

[image]

brentk7, 2024-07-08:
Does anybody have any pointers on what I can do to troubleshoot or help fix this issue?

rstrouse, 2024-07-08:
I will try to get to it later this week.  Unfortunately, I do not have one of these boards so I need to see if Sequent changed the registers for the inputs in their code.

brentk7, 2024-07-08:
Much appreciated - please let me know if you need any info, logs or testing.

rstrouse, 2024-07-26:
Update your relayEquipmentManager.  I checked all the registers and the registers seem to match the 4rel4in from the Sequent codebase.  It should be reading the correct i2c registers.

brentk7, 2024-07-26:
I have just pulled the latest version from github and am still getting the same error

rstrouse, 2024-07-26:
Have you installed the cli from sequent to see if it works?

brentk7, 2024-07-26:
Yes - works fine from sequent cli and from sequent Node Red

rstrouse, 2024-07-26:
I don't know why I was thinking that it was not able to retrieve the values from the device.  As it turns out even your log shows a successful read of the register.  However, the caught exception below reflects that it was not able to find the channel value for the input.  I realized this last night and let it run all night pulling mock values.

Update your REM and let me know if it is doing its thing with the hardware.

brentk7, 2024-07-27:
Thanks that seems to have fixed it. I'm now able to change panelMode, but still can't stop / start my pump from this. Have posted an extra question [here ](https://github.com/tagyoureit/nodejs-poolController/discussions/508) as i think that is more relevant place for others to look for a reply.

closed 2024-07-27 by brentk7