rstrouse/relayEquipmentManager #76: 4-20mA pressure sensor

dakowals, 2024-07-03:
Anyway to setup a 4-20mA pressure sensor with the SM IND board input.  I can acquire the signal, but can only use voltage when setting up a generic device for filter pressure.

tagyoureit, 2024-07-07:
Which pressure sensor and ph and orp sensors do you have?

dakowals, 2024-07-07:
The pressure sensor is a Balluff BSP0016.  4-20mA output.  The PH and ORP are GF Signet 2700 series probes going to the GF 9900 series transmitters with 4-20mA output
[image]
[image]
[image]

rstrouse, 2024-07-07:
You can try to feed the mA from the IND to a generic IXIAN pH transmitter to see if it translates the values properly into pH.  The calculation for that device is below to convert mA into pH.
```
14 * ((device.values.adcValue - 4) / 16)
```

dakowals, 2024-07-08:
I have added them as Generic devices, but cannot see them in poolController.  The data is live for the generic device, but does not carry over.  See attached setup:
[image]
[image]
[image]

Also see how I have attached the filter pressure.  It displays, but the range is wrong:
[image]
[image]

Actual pressure is about 8.8 PSI

rstrouse, 2024-07-08:
A couple of things.  First these are definitely not EZO probes so njsPC will not be able to simply pull the data and set up an output.   You need to select other as the type of probe then feed it to the REM Chem controller through the feeds.  Is the pH value correct?

Also, there is not an 4-20mA ORP probe in REM yet as I have no idea what the formula translation is from mA to mV on the probe.

Finally, you will need to create a generic pressure device and feed the adcValue to it for the pressure but I suspect this too does not have a corresponding mA to psi translation.

dakowals, 2024-07-09:
The scale for the PH I can set to what ever I would like.  PH is (0PH to 14PH) for 4-20mA.  ORP is (-2000mv to +2000mv) for 4-20mA.  PH is now reading, but the scale is off.  ORP is not since there is no scalar for mv.  When I add in the pressure sensor, the only option for input is voltage.  I cannot select adcValue.
[image]

tagyoureit, 2024-07-10:
I just put in some screenshots for hooking up a pressure transducer in a separate thread.  https://github.com/tagyoureit/nodejs-poolController/discussions/995#discussioncomment-10004164

dakowals, 2024-07-10:
The method for adding is exactly how I added it to my poolController.  The problem is the sensor is 4-20mA current and not voltage.  There is data coming through, but it stays at 4.8psi.  There is no way to scale the voltage for a current.

dakowals, 2024-08-20:
Is there any way to map the current scale to voltage?  Is there another way to display my PH and ORP values in my poolController main page as just a value from REM?