rstrouse/relayEquipmentManager #87: BUG - Relays are switched.

spifftek70, 2024-12-29:
R6 and R7 are switched.  When I use REM to turn on R6, the light for R7 comes on the Eight Relays 4A/120V 8-Layer Stackable HAT board.  and vice-versa.

tagyoureit, 2024-12-30:
Can you confirm which version FW you have?

spifftek70, 2024-12-30:
> Can you confirm which version FW you have?

```
megabas v1.2.6 Copyright (c) 2016 - 2023 Sequent Microsystems

This is free software with ABSOLUTELY NO WARRANTY.
For details type: megabas -warranty
```
and

```
8relind v1.1.1 Copyright (c) 2016 - 2020 Sequent Microsystems

This is free software with ABSOLUTELY NO WARRANTY.
For details type: 8relind -warranty
```

tagyoureit, 2024-12-31:
1.  What version hardware is the 8relind board?  Looks like they are up to v6 per the picture on their website.
2.  Do you have the latest FW? 
3.  If you use the [command line)[https://github.com/SequentMicrosystems/8relind-rpi/tree/main/python] and trigger the relays are they lighting up the correct leds?

Reason I specifically ask about the last one is Sequent has been known to either have leds swapped/reversed/incorrectly screen printed.  They have also been known to swap registers between hardware versions.

spifftek70, 2024-12-31:
1. the newest version from their site. 
2. yes... yes I do.
3. You might be on to something.  The relays for 6 & 7 were reversed when I tested them with this script:

```
import lib8relind
import time

# Stack level (0 by default)
stack_level = 0

# Number of relays on the board
num_relays = 8

# Iterate through each relay
for relay in range(1, num_relays + 1):
    # Turn the relay ON
    lib8relind.set(stack_level, relay, 1)
    print(f"Relay {relay} is ON")
    time.sleep(3)  # Wait for 3 seconds

    # Turn the relay OFF
    lib8relind.set(stack_level, relay, 0)
    print(f"Relay {relay} is OFF")
    ```

spifftek70, 2024-12-31:
BTW, do you like my 3D printed DIN bracket for the Stack?  

[image]
[image]

spifftek70, 2024-12-31:
It's a hardware error. The relays are correct, but the LEDs are lit wrong.

tagyoureit, 2024-12-31:
I was just taking a look at the registers.  So you don't need anything changed?  It all works except the wrong LED's light up for relay 6 & 7?  

And just because it's fun to find more issues...
1.  The screening on the front indicates that all relays except 4 are 5A/48VAC/DC while relay 4 is 3A/48VAC/DC.  However, the rear screen printing shows relays 1/3/6/8 as 5A/48V while 2/4/5/7 are 3A/48V.
2.  "oposite"

[image]

Also, nice job on the DIN rail bracket.  My only comment is that the RPi stacks generally stay cool via passive heat distribution.  I can't see the bottom, but I'd monitor the temps and if they start to crawl up you may want to add some more passages for air ventilation.

spifftek70, 2024-12-31:
I actually haven't plugged it in to the pool components yet.  I'm waiting for someone with much more experience to find these types of important details.  😀

spifftek70, 2025-01-08:
https://github.com/SequentMicrosystems/megabas-rpi/wiki/User's-guide-4.2#errata

"The silkscreen of the Version 4.2 of the Building Automation Card has an error on the function of the DIP switches for selecting the input type. The correct silkscreen is shown below. It applies to each of the eight analog/digital inputs on the left side of the board."

shinerblue, 2025-01-11:
> It's a hardware error. The relays are correct, but the LEDs are lit wrong.

I can confirm the same on my V6 HAT as well.