rstrouse/relayEquipmentManager #88: New REM installation on a different RPI--can I copy

davthomaspilot, 2025-01-06:
Can I copy relayEquipmentManager/data/controllConfig.json into the same spot on a new REM installation to eliminate the need to configure the new install?

tagyoureit, 2025-01-07:
Yes, just make sure REM is stopped (in the new location) when you copy the file or it will be overwritten.

davthomaspilot, 2025-01-07:
So, I installed REM.  It's now on the same RPI image as I am using for njsPC and dashPanel.  Previously, njsPC and dashPanel were running on an rpi in my equipment closet and REM was running on a different RPI in my pool equipment area.  Now I plan to run everything on the RPI in the equipment area so I can wire directly RS485 from that RPI to the OCP (and eliminate the need for the Pentair wireless link).

Did the REM install based on the instrcutions on your wiki, then just copied the data folder.  Got this herror 

```
linux
[07/01/2025, 11:15:53] info: Starting up SSDP server
[07/01/2025, 11:15:53] info: Initializing GPIO Pins 3
[07/01/2025, 11:15:53] info: Configuring Pin #37 Gpio #26:in on Header 1 Edge: both. {"activeLow":false,"reconfigureDirection":false}
[07/01/2025, 11:15:53] info: Server is now listening on 0.0.0.0:8080
Error: EINVAL: invalid argument, write
    at Object.writeFileSync (node:fs:2367:20)
    at exportGpio (/home/pi/relayEquipmentManager/node_modules/onoff/onoff.js:18:8)
    at new Gpio (/home/pi/relayEquipmentManager/node_modules/onoff/onoff.js:172:36)
    at GpioController.initPin (/home/pi/relayEquipmentManager/gpio/Gpio-Controller.ts:103:36)
    at GpioController.initPins (/home/pi/relayEquipmentManager/gpio/Gpio-Controller.ts:142:28)
    at GpioController.init (/home/pi/relayEquipmentManager/gpio/Gpio-Controller.ts:22:14)
    at /home/pi/relayEquipmentManager/app.ts:22:38
    at processTicksAndRejections (node:internal/process/task_queues:95:5)

REM dies.

Old REM rpi was a 3B, Bullseye, new one is RPI 4, Bookworms

rstrouse, 2025-01-07:
You need to change the board setting to the bookworm setting in the configuration.  This is on the general tab.  The geniuses at RPI decided to break every piece of code written for it that accesses GPIO by changing the addresses of the pins.  There's more to it than that but the results are as described.

davthomaspilot, 2025-01-07:
general tab of what?  I can't connect to the REM server via port 8080 since REM dies.    Is it the json file somewhere I can edit?

davthomaspilot, 2025-01-07:
defaultController.json has:

defaultController.json:  "controllerType": "raspi",

Perhaps change that to 

raspi-4b-bookworm           ?

( see files raspi.json and raspi-4b-bookworm.json in pinouts folder)

davthomaspilot, 2025-01-07:
That did not work.  Same thing happens

davthomaspilot, 2025-01-07:
Started over with REM install.  This time I did not copy the data folder.  I found the General tab and changed to Bookworm.

I'll worry about get REM configured once I get njsPC/dashpanel running and connected to OCP RS485.

closed 2025-01-07 by davthomaspilot