rstrouse/relayEquipmentManager #90: GPIO26 isn't sensed correctly on RPI4

davthomaspilot, 2025-01-08:
I use pin 37, GPIO 26 as a flow switch input.  It worked fine on RPI 3B+, Bullseye:

[image]

But, the Pin State is not sensed correctly nor is recognized in njsPC.

I can use pigpio tro check the state of GPIO26--when flow is on, GPIO26 is high, when flow is off, GPIO26 is low.

If I click the Inverted check box, njsPC thinks there is flow, but it doesn't change when flow goes off.  It's like the GPIO that's being monitored by njsPC is incorrect.

Here's the RPI 4 pin-out--but it's the same as it was in earlier pi versions:

[image]

davthomaspilot, 2025-01-08:
Looks line line 690 in relayEquipmentManager/pinouts is incorrect.

Most gpioID differ from the BCM number by 512--I'm guessing that's because of different offset for GPIO memory mapping RPI4 versus previous versions.

gpioD was 528-I think it should be 538 (26+512).  I made that change and it fixed the issue for me.

[image]

There other instances where the difference is not 512, but I'm too ignorant to know if maybe that's actually correct for some GPIO.

davthomaspilot, 2025-01-08:
_here other instances where the difference is not 512, but I'm too ignorant to know if maybe that's actually correct for some GPIO._

Eg, line 684:
[image]

Like someone was going through and adding 512 to the BCM # but missed BCM 16?  Or, maybe my ignorance...

tagyoureit, 2025-01-14:
I've been looking into this.  I need to dig deeper.  I'm not seeing why the gpio's are numbered the way they are.  And it isn't clear to me if they will change between different hardware (I think it might) or not.  @rstrouse if you have time, can you shed some light here?

davthomaspilot, 2025-03-28:
I thought it was fixed.  

It's bee winter time, so I've not been paying attention to njsPC or REM.  Today I see that the REM pinout configuration has been lost.

Worse, when I try to reconfigure (starting with the GPIO for the flow switch) I get:

[image]

I haven't updating anything on the rpi, but it has rebooted.

What now?

davthomaspilot, 2025-03-28:
And, when I try to restart REM, it crashes:

Using username "pi".
Authenticating with public key "rsa-key-20160906"
Linux PoolController 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Mar 28 14:15:05 2025 from 192.168.0.6
pi@PoolController:~ $ cd relay*
pi@PoolController:~/relayEquipmentManager $ /usr/bin/nod dist/app.js
-bash: /usr/bin/nod: No such file or directory
pi@PoolController:~/relayEquipmentManager $ /usr/bin/node dist/app.js
linux
[28/03/2025, 14:17:06] info: Starting up SSDP server
[28/03/2025, 14:17:06] info: Initializing GPIO Pins 2
[28/03/2025, 14:17:06] info: Configuring Pin #37 Gpio #26:in on Header 1 Edge: both. {"activeLow":false,"reconfigureDirection":false,"debounceTimeout":1}
[28/03/2025, 14:17:06] info: Server is now listening on 0.0.0.0:8080
Error: EINVAL: invalid argument, write
    at Object.writeFileSync (node:fs:2367:20)
    at exportGpio (/home/pi/relayEquipmentManager/node_modules/onoff/onoff.js:18:8)
    at new Gpio (/home/pi/relayEquipmentManager/node_modules/onoff/onoff.js:172:36)
    at GpioController.initPin (/home/pi/relayEquipmentManager/gpio/Gpio-Controller.ts:103:36)
    at GpioController.initPins (/home/pi/relayEquipmentManager/gpio/Gpio-Controller.ts:142:28)
    at GpioController.init (/home/pi/relayEquipmentManager/gpio/Gpio-Controller.ts:22:14)
    at /home/pi/relayEquipmentManager/app.ts:22:38
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
pi@PoolController:~/relayEquipmentManager $

I could really use some help.  Should I try going back to the GPIO definitions in the distribution that we thought were wrong for an RPI 4?

davthomaspilot, 2025-04-03:
I guess I'll go back to RPI3.

If anyone is reading the REM issues anymore, maybe you should mention it doesn't work on RPI4.  Maybe save some like me some trouble.

davthomaspilot, 2025-04-03:
Maybe it's not RPI4, but just newer debian release.  I found this:

https://www.thegoodpenguin.co.uk/blog/stop-using-sys-class-gpio-its-deprecated/

which seems to say writing to /sys/class/gpio  (like is done is onfof.js in REM) is obsolete.

Is there a plan to fix this, or should I trying hacking something myself?  Or, dig up a legacy OS version and use that?

davthomaspilot, 2025-04-04:
I gave up and went back to an RPI3 on Bullseye.  Everything is working ok.

One approach to get REM working on Bookworm and RPI4 and later would be to use the pigpio library for the GPIO input and output configuration.  I've used that library with a socket connection to a daemon (pipiod) so that root access is only needed for the daemon--not the client applicaiton.

If you guys want to address the GPIO issues using pigpio, let me know.   I can write to an interface we define (maybe the interface already exists), but I'd want to "kibitz" before embarking.

tagyoureit, 2025-06-13:
Nope... have not had a chance to look at this.  Don't have any ETA for you, either.

essbennett, 2025-06-25:
This may be obvious to everyone, but I was having trouble setting up REM. I am running on a raspberry pi 4B and Debian 12.11. I had selected Raspberry Pi instead of Raspberry Pi 4B Bookworm OS for the board. Once I selected the correct board I was able to assign the pins and triggers. I did find a few places in both raspi.json and raspi-4b-bookworm.json "pins": [] where I changed the gpioId to match the id: (raspi) or gpioId+512 (raspio-4b-bookworm)  as davethomaspilot first noted ( I would not have figured out the 512 offset without his posts). Now it is working like a champ, thanks so much for this great program!

essbennett, 2025-07-15:
> I'd like to know if it continues to work for you.
> Mine worked originally, but then I started getting the same errors, even
> with the offset thing.  I don't know if that was after a reboot, or what.
> I could never get it working again.
> The way the code uses GPIO is now deprecated.
> At some point, if the guys don't update to something support on newer RPI
> and Debian versions, I will hack the code to use a library like Pigpio.
> But, I'm not at all good at JavaScript.

So far so good. I have done multiple restarts as I was working on an issue
with the pool controller module. I'll let you know if I have any issues.