rstrouse/relayEquipmentManager #91: REM not providing data

ddyer8, 2025-03-09:
Yesterday it appears I lost my REM feeds after a power-cycle on the equipment.  I tried updating REM from v8.0 to v8.1.  Running v8.04 of pool controller.

[image]

If I try to interact with i2c devices, I'm told the bus doesn't exist.  I was tempted to delete the i2c bus in REM, but was reluctant to reassign all my devices.

[image]

### REM error log contents:
TypeError: Cannot read properties of undefined (reading 'split')
    at AtlasEZOpmp.getPumpVoltage (/home/pi/relayEquipmentManager/i2c-bus/AtlasEZO.ts:865:34)
    at AtlasEZOpmp.getDeviceInformation (/home/pi/relayEquipmentManager/i2c-bus/AtlasEZO.ts:873:13)
npm notice 
npm notice New major version of npm available! 9.5.0 -> 11.2.0
npm notice Changelog: <https://github.com/npm/cli/releases/tag/v11.2.0>
npm notice Run `npm install -g npm@11.2.0` to update!
npm notice 

/home/pi/relayEquipmentManager/i2c-bus/AtlasEZO.ts:865
            let arrDims = result.split(',');
                                 ^
TypeError: Cannot read properties of undefined (reading 'split')
    at AtlasEZOpmp.getPumpVoltage (/home/pi/relayEquipmentManager/i2c-bus/AtlasEZO.ts:865:34)
    at AtlasEZOpmp.getDeviceInformation (/home/pi/relayEquipmentManager/i2c-bus/AtlasEZO.ts:873:13)

ddyer8, 2025-03-09:
I get the same error as #13, but all my I2C comm seems lost.

tagyoureit, 2025-03-10:
You can use a backup if you have one, but something got corrupt and you'll need to recreate the bus and devices.

tagyoureit, 2025-03-20:
Do you have REM running when you use i2c detect?  If so, stop it first.

ddyer8, 2025-03-21:
REM is off. Is it time for me to try a fresh install? Do you publish/recommend a docker container for rpi4?Thank you,Dustin

tagyoureit, 2025-03-23:
There is [docker](https://github.com/tagyoureit/nodejs-poolController/wiki/Docker) for njsPC and dP.  REM needs direct control over the hardware.

tagyoureit, 2025-04-15:
REM is unable to find ` i2c:1:2:in0_10.2`.  This is input 2 of a 0-10 input.  If that's where your flow switch is, then yes.

ddyer8, 2025-04-15:
I’ll re-check, but that’s where my flow switch is. Do I need to configure a feed out of REM? I didn’t see a set of options or generic device that seemed appropriate.

ddyer8, 2025-04-24:
I'm not 100% sure what the root cause was.  The fan on my Sequent SmartFan was seized.  I replaced and REM is providing feeds again.

closed 2025-04-24 by ddyer8