rstrouse/relayEquipmentManager #93: Unable to feed temperature with Home Automation v4 HAT

derickcassidy, 2025-05-04:
Good morning

I have a Home Automation v4 HAT from Sequent.  It is running firmware 1.34.  HW version 4.0.  I have a fresh install of the latest REM on a fresh OS install on a Raspberry Pi 5 Model B Rev 1.0

After cloning and building REM, I can launch a browser to localhost:8080 and see the server running.  I added an I2C bus, and can see the the HAT on I2C Device Definition for Address 40 - 0x28.

I hooked up a thermistor to ADC3.  I enabled the Analog input, called it ambient and saved the device.  I can see the thermistor is being read and has 1.41v across it.  Holding the thermistor reduces the voltage as expected, so I'm confident the input is reading correctly.  

[image]

I created a generic device called ambientGenericDevice as type 10k Temperature Probe, set the input as Raw Value, 12-bit Input and 10k thermistor table. I saved the device. 

[image]

Now here's the issue:  When I go to Feeds, and attempt to map ADC3 to the generic device, I never see the input as a "Send Value" in the Internal Devices drop down.  It is empty.  

[image]

 Does anyone have any idea what might be happening?  I read the previous issues around the Home Automation HAT and the temperature/relay states.  But I have confirmed that I am running latest, and the referenced pulls have been merged.  I can see data being received in the ADC3 input.  I can see the proper Voltage being referenced.  But, I cannot see the input in the Feeds section to be able to map it to the generic device.

I feel like I am missing something simple. Any help is appreciated.  I've attached a silly log of startup, clicking through the generic devices, the ADC3 input, and the attempt to save the feed. 

Cheers.

[silly.log](https://github.com/user-attachments/files/20027791/silly.log)

derickcassidy, 2025-05-05:
Bit of an update.  I was able to check the SequentIO.ts file in the i2c-bus directory.  Many of the i2c addresses were incorrect for the Home Automation HAT.  By changing to match the output of 'ioplus 0 board' to have the proper addresses, I can now see and map temperature correctly.  

However, relays are not showing up in the list of devices that can be mapped to the pool controller and I think that is because of the getDeviceDescriptions() referenced in the other thread.  Will head down this path today and see if it doesn't address the issues.

derickcassidy, 2025-05-06:
This is solved.

getDeviceDescriptions() had to be updated to include any enabled relays. 
Additionally, the getDeviceState() and setDeviceState() functions needed to be copied over.

Once these were added, and the proper i2c registers updated, all is working.

closed 2025-05-06 by derickcassidy

tagyoureit, 2025-05-09:
Thanks for digging in.  Feel free to open up a pull request.