rstrouse/relayEquipmentManager #94: REM doesn't work on bookworm, at least on some GPIO

davthomaspilot, 2025-05-17:
See my other recent issue for details.

I thought it was RPI4 versus RPI3, but actually bookworm on RPI3B+ has the same issue.  But, it's fine on Bullseye, as imaged from the RPI imager as of 5/15/2025.

The issue may be specific to the pins I'm using for flow sensor, ph and ORP sensors, and chlorine/acid pumps.

The fix may be as trivial as changing a file that maps GPIO to address space.  I've "fixed" by editing a file, but for some reason the fix quits working.

Will this ever be addressed, or generally, will REM be maintained in the future?

jaysalazar, 2025-05-20:
Do you have the following setting selected?
[image]

bruajr, 2025-05-20:
> Do you have the following setting selected? [image]
I was having the same issue and was about remove everything and start over with Bullseye. This fixed it! Thanks!

davthomaspilot, 2025-05-23:
Yes

davthomaspilot, 2025-05-23:
For details ,  please read issue #90.  It probably works on most of the GPIO, but it looks like there was a mistake in a file that maps GPIO to address space.

I made a change to that file that "fixed it" for a while, but it later started having the same issue.   That's when I gave up and went back to Bullseye