rstrouse/relayEquipmentManager #102: REM Heap Size Issue

Exit2Studios, 2025-07-11:
[image]

As you can see from the screenshot, REM is using a ton of Heap memory and causing repeated crashes.  I've tried to add a memory limit within the ecosystem.config.js file, but that doesn't stop the issue.

Any ideas what could be causing this or best way to remedy it?  I'm currently running this on an 8GB Raspberry Pi 5, so I shouldn't be having memory issues.

Exit2Studios, 2025-07-12:
I reinstalled the latest version on existing hardware that was working a few years ago. I wiped the as card and started over today and the memory creep seems to have stopped so I have no idea what was causing it. I’m building things back out systematically and keeping an eye on heap and memory usage and once I’m sure I’m free of crashes and reboots, I’ll put it into production. I am getting a strange warning that REM is looking for my MegaBas card on i2c port 2, which it isn’t on, so not sure what to do about that. It’s just warning, but I hate filling logs with noise.

tagyoureit, 2025-07-12:
Try to delete anything on the 2nd port if it doesn't exist.  Or you can manually try to delete invalid entries from `data/controllerConfig.json` but that really isn't recommended.  Open up a different issue for that if you need help.