rstrouse/relayEquipmentManager #106: How to integrate MEGABAS 1-Wire CLI temperature readings into REM?

SeaFox63, 2025-08-03:
Hi @rstrouse,

I’m using the Sequent Microsystems MEGABAS building automation HAT with REM on a Raspberry Pi 4 running Bookworm. The MEGABAS board has a 1-Wire sensor interface but it can’t be read via /sys/bus/w1/devices, and so can't be seen in REM.

Instead, the MEGABAS 1-wire bus sensor readings come from the CLI using Sequent's own software, eg:

     megabas 0 owbtrd 1  → returns temperature in °C

I’d like REM to pull temperature values from the BAS HAT using this megabas command. I've tried adding a megabas subtype to temperature.json. But all I succeeded in down was breaking the UI causing the MEGABAS board to disappear. Clearly I’m not doing this right. 

Is there any supported way to do way to get temperature values from the the MEGABAS board into REM that I can then use in NJSPC?

Thanks so much for your help — it really is an amazing project!

Charles

tagyoureit, 2025-08-07:
The MEGABAS does include a standard 1-Wire interface. According to Sequent Microsystems’ documentation, and previous experience, it allows connecting DS18B20 sensors in parallel, which should show up under /sys/bus/w1/devices if the 1-Wire kernel modules and GPIO overlay are correctly configured.

Could you double-check that 1-Wire is enabled in /boot/config.txt (look for dtoverlay=w1-gpio) and that the kernel modules (w1-gpio, w1-therm) are loaded?

Sequent’s CLI just provides an alternative method to access sensors, but the standard kernel interface should still be usable and compatible with REM.

SeaFox63, 2025-08-17:
Thanks for getting and sorry for taking so long to reply. 

Double Check: I have checked config.txt and lsmod. The interface is enabled and the kernel modules are loaded. 

Symptoms: If I connect the 1-wire sensors to gpio4, I am able to get temperature values from the raspi command line, and also through REM. But if I connect the 1-wire sensors to the BAS card connector, the sensors are not seen either by the raspi command line, or by REM. 

Sequent advice: I asked Sequent what was wrong. Their reply was: "_The one-wire bus exposed at the card connector is connected to the local processor. The one read by Linux is connected to Raspberry GPIO4 (2x20 pin header on 7) ... the local processor is accessible through the I2C port but has no kernel driver ... so, the fastest way to make it available to the kernel is to use GPIO4_" 

My project: I am working on a integration class to wrap the megabas <address> owbtrd <sensor number> in some code to allow REM to read it. Some progress, but slow progress. Any advice would be wonderful!

Best, Charles